kipr/libwallaby #11: IMU calibration

jsoutherland, 2015-11-05:
For the:
* accelerometer
* gyrometer
* magnetometer

JSRGFJZ, 2016-02-22:
where is exactly the code

bowbahdoe, 2016-02-22:
@JSRGFJZ There is no code yet, this is a task that needs to be done.

JSRGFJZ, 2016-02-23:
Well , do you guys have some solutions that can promote the accuracy

jsoutherland, 2016-02-23:
Hi @JSRGFJZ 

Sorry this has been on our ToDo list.  I will try to get to it soon but I can't promise it will be in the next update.

In the past, we did not have magnetometers and provided bias calibration for the accelerometer and gyro.

You can keep the robot still for a time (30 seconds?)  and fill arrays with sensor readings from each.  Then you can average them to solve for a bias.   

The goals would be...

Accelerometer:
* x = 0
* y = 0
* z = -1024

Gyro
* x = 0
* y = 0
* z = 0

So if your Gyro average was   ``<x,y,z> = <-10,5,2>`` and the expected average was ``<0,0,0>``
then you would add ``<0 - -10, 0 - 5, 0 - 2>``  or ``<10, 5, 2>`` to future gyro readings on that board 

Accelerometer is the same, except you use the goal/expected vector of <0, 0, -1024> when computing the average error.

Magnetometer is trickier since it involves rotations.  Were you planning on using that one?

Zacharyprime, 2019-10-03:
I just added Accelerometer and Gyroscope calibration functions and the compass was done by Dr. Fagg if I am not mistaken.

For documentation purposes, Dr. Faggs compass calibration essentially attempts to round a ellipsoid in order to produce a more accurate output. He uses measurements of 8 axis per plane to run his calculations.

The Accelerometer and Gyroscope are calibrated using low-pass filtering. Maybe in the future other filters could be added such as Kalman filters, fourier series removal of noise, etc. 

For now the IMU on the Wombat is good enough that hopefully the calibration functions won't need as much use as on the Wallaby.

closed 2019-10-03 by Zacharyprime