kipr/libwallaby #54: Support higher camera resolutions

navzam, 2016-01-30:
Currently we only support 160x120. It would be nice to support at least 320x240.

EugeneDMyers, 2020-01-29:
The code does support this (320x240) and higher resolutions.  However, going to 320x240, for example, will take 4x processor power.  Will have to remove the restriction in the code and test to see if the processor can handle the higher resolution, which should not take long.  Should also do this test under load as this will increase the USB bandwidth requirements by 4x as well.

EugeneDMyers, 2020-02-01:
If you want test higher resolutions, comment out lines 31-33 in camera.cpp, that disables the check in camera_open_at_res().  You can also test using, camera_open_device_model_at_res() and camera_open_device() to control the resolution.

closed 2020-03-10 by prithvirajkadiyala