kipr/libwallaby #55: Crash if camera is already open

navzam, 2016-02-04:
If the camera is being used by another source, camera_open() usually crashes the user program.

Zacharyprime, 2020-01-30:
This is confirmed to be an issue on the Wombat. 
The program will return with NULL instead of 0 and crash if the camera is open in Botui.

EugeneDMyers, 2020-02-01:
Actually, the 0 return is normal (and expected behavior) when the camera cannot be opened.  The user program should check to see if there is a 0 return when the camera cannot be opened because BOTUI has control of the camera.  

A crash can be prevented by camera::Device::update (camera.cpp) check to see if the camera is opened by calling camera::Device::isOpen().  That is a best a bandaid solution as the code in camera.cpp is a hodgpodge of support for non-opencv and opencv cameras.

EugeneDMyers, 2020-02-01:
Also, in this case the user program will loop waiting for a frame to appear.  They still need to check for this case.