kipr/libwallaby #59: camera_load_config()

symbioticcomputing, 2016-02-15:
camera_load_config() does not seem to be working, as it always returns 0 and does not load the requested configuration.

For our test conditions, we have a single color configuration file ("Canberra.conf") specified in the GUI that sets up a single channel (I am not sure where this is stored in the file system).  This configuration file was not designated as the default (and no others were).  camera_load_config() returns a zero (presumably indicating an error) - and any call to get_object_count(0) results in a printed error message that the "specified channel must be between 0 and some-large-number."  This suggests that a configuration had not been set up.  

Furthermore, when Canberra.conf is configured as the default configuration in the GUI, camera_load_config() still behaves the same, however get_object_count(0) does not fail.

//////////////////////////////////////////////////////////////////////////////////////////////////////
#include <kipr/botball.h>

int main()

{
  int res;

  res = camera_open(LOW_RES);
  printf("Open result: %d\n", res);

  // Does not seem to work: always returns 0
  res = camera_load_config("Canberra.conf");
  printf("Configuration result: %d\n", res);

  int object_count;
  point2 p;

  while(digital(1) == 0)
  {
    // Address buffering issue
    res = camera_update();
    res = camera_update();
    res = camera_update();
    res = camera_update();
    printf("Update result: %d\n", res);

    object_count = get_object_count(0);
    printf("Number of objects %d \n",object_count);
    if(object_count > 0){
      p = get_object_center(0,0);
      printf("Largest object: %d %d\n", p.x, p.y);
    }

    msleep(500);

  }

  camera_close();
  printf("Done...\n");
  msleep(500);
  return 0;
}

navzam, 2016-02-15:
@symbioticcomputing - Can you try removing the file extension from the camera_load_config() call?

`res = camera_load_config("Canberra");`

Right now, the function call will always add the extension for you. We should make this more lenient (so it can determine the presence of an extension), as well as document the expected input.

symbioticcomputing, 2016-02-15:
This solved it. 

Right now, the best documentation that we have says that the user needs to add the extension.  

Also - the error message from get_object_count() is a bit misleading.  Maybe something along the lines of "channel %d not defined in configuration" would be more informative.

navzam, 2016-02-15:
Great. Thanks for the feedback.

closed 2016-02-15 by navzam