kipr/libwallaby #69: wait_for_light() May Make Motors and Servos Behave Strangely

b4hofer, 2016-04-12:
When using the wait_for_light function from libwallaby, our motors and servos would sometimes behave very strangely (motors not moving at full speed, servos moving without instruction). It doesn't happen all the time, and we are unfortunately not sure how to reproduce it.

jsoutherland, 2016-04-12:
@b4hofer Thank you for reporting this issue.  I will look into it.   If you happen to reproduce it at some point with a small set of code that you don't mind sharing, that will definitely help.

Zacharyprime, 2020-01-30:
The wait for light function has been updated. There is no experimental behavior found that appears to affect the motors. 

It is unlikely that people will use the function if motors/servos are moving anyway as it's really only useful for starting the match.

The function will however behave similar to msleep() or wait_for_button() in regards to the motors and servos so it can be used that way. There are no calls to alter motor/servos in the function (as of now).

closed 2020-01-30 by Zacharyprime