kipr/libwallaby #100: Gyro not reading out accurate values

prithvirajkadiyala, 2020-01-29:


Zacharyprime, 2020-01-30:
The gyroscope is reading out precise values now. There is a small accuracy problem but the pull request I created solves it to the point where the gyroscope is usable. The patch name is ZacharyGyro-Patch-A.

Sometimes there is a spike in the data that I can't find the source of and when integrating the data to get absolute position there is a +/- 10 degree inaccuracy that can be minimized slightly by adjusting sample frequency. 

For botball purposes, I think it will work.

Zacharyprime, 2020-10-08:
I wanted to add clarification to the current state of the issue.

The output of the gyro has spikes that happen seemingly at random. They also are not regular.
For example, the output shown at the bottom required me to run the program 3-4 times before it showed a problem.

I still haven't found a identifying pattern to the behavior, the only information I've found is that it's been the high register that is incorrect in every test I've done (resulting in a large spike since it's most significant). The odd part is that the IMU isn't read via a bus like you would expect a bit flip error like this to happen, it's done using SPI communication... 

I do believe it would be possible to move the IMU to be handled by the Pi (just like all the non-analog devices). This would require an expensive rework and would need to be packaged with other major design changes to be worth it. I wouldn't be surprised if the STM32 was the cause of the issue, since moving the SD reader and USBs to the Pi fixed similar data corruption issues too.

The "angle" number is a trapezoidal sum with the data filtered using a time-domain high pass filter (there may be a moving average filter, I don't remember if I had it enabled that run).

[image]

Here is a schematic of the IMU in case anyone does not have it:
[image]

The firmware and libwallaby code seem fine to me, the schematic also doesn't show anything that would cause this error (in theory since this is EAGLE), maybe there is a manufacturing issue or something? I'm at a loss of ideas at this time.

Zacharyprime, 2022-04-22:
As far as I can tell, the gyroscope is outputting correct data, but when I tried to integrate the data to make a gyroscope function, it was not accurate.
I ran some experiments to try and figure out what exactly is causing the inaccuracy, but every experiment I run shows that the data is good.
My current thought is maybe the sample rate isn't set high enough, but I've been working on higher priority tasks so I haven't looked into the sample rate yet.
https://docs.google.com/spreadsheets/d/1fwIDw7y76WylASVldnxO-bq3UiNzY-dxusHyrINDscI/edit?usp=sharing

The "Linear Speed Test" tab contains my data from the best experiment I ran.

Zacharyprime, 2022-05-16:
I was looking around for something else and found this.
https://invensense.tdk.com/download-pdf/migrating-from-mpu-9250-to-icm-20948/

I need to check what is actually on the board because I thought we used the 6050, but my documentation says we are using the 9250. I am probably just mixing up another project I used the 6050 for, I trust the documentation more than my gut in this case.

Zacharyprime, 2022-08-16:
Braden and Noble (botball team) have worked on this quite a bit. 
If @bmcdorman could fill in where we are at on this.