kipr/libwallaby #107: Camera live crashes botui

prithvirajkadiyala, 2020-02-26:
While testing the camera live view, BOTUI crashes when we try to go back.
This issues seems to be there after the merging of the pull request #106 

Look into what resources are being consumed for the camera and what is being returned from the functions while calling any of the camera functions.

EugeneDMyers, 2020-02-26:
did this occur before pull request #106?

prithvirajkadiyala, 2020-02-26:
No it did not. That's why I had to crate one now.
Tested with the earlier versions and seems to be working fine.

EugeneDMyers, 2020-02-26:
Make sure that BOTUI is rebuilt using the new libwallaby header files.  There have been some changes in some of the defaults that might affect BOTUI.

That was one of the problems last year.  Because of personnel turnover, several actions did not get carried over. I've been reading my notes and I need to look tonight at some of the fixes I made locally to BOTUI as well.  Those changes I never got around to pushing them out.

EugeneDMyers, 2020-02-26:
Another issue, which you can check, was making the correct version of BOTUI is executing.  Last year, the build system (make install) placed BOTUI in /usr/local/bin (where botui_loop.sh is located, which makes sense.  The install system was placing a copy in /usr/bin, which really confused things.

EugeneDMyers, 2020-02-26:
Here is an excerpt from last year:

"Also, when you create the next release, load botui last. That will make sure that every thing gets copied.  The reason that the boot does not work is that the script needs to run in the background so when botui crashes, it doesn’t bring it down.  I’ ll let you figure out how to do that."

prithvirajkadiyala, 2020-02-26:
Although we do not exclusively check for the loop file. We do make sure to delete if there is any during the update via USB or Online Updates via the script, which results in cleaning out the old version completely.

I remember having some issues with that during the Wombat-v25.4 when we were trying to implement something and it just wouldn't update(versions) regardless of the number of times we did the update.

EugeneDMyers, 2020-02-26:
Generally, the loop sh keeps restarting BOTUI and linux prevents the BOTUI file from being overwritten while BOTUI is executing.

So, for any update, BOTUI has to not be executing, which makes it difficult when you want to do an update from the GUI.

EugeneDMyers, 2020-02-27:
Checked my wombat and found two different botui's one under /usr/bin and the other under /usr/local/bin

both different

prithvirajkadiyala, 2020-02-27:
In that case I would recommend doing the USB update or Online update if you're plugged via Ethernet. There is a line in the update code that solves the multiple versions issue.

EugeneDMyers, 2020-03-01:
A resolution to this issue is in the pull requests for libwallaby

prithvirajkadiyala, 2020-03-02:
Deleting the line where we force "model=black_2017" doesn't open the camera.
I will look into what more changes will fix or break stuff.

EugeneDMyers, 2020-03-02:
Could you explain further?

prithvirajkadiyala, 2020-03-02:
After merging the new pull request. Camera live wasn't showing up.
So I was looking at the changes made and inserted **model=BLACK_2017** at line 320. After inserting the line back the Camera live showed up. Testing to see what else might have broken it.

And the Back or Home button still crashes the UI.

I am trying to test camera at the Home->Motors and Sensors->Camera

https://github.com/kipr/libwallaby/blob/ab47eec70b9228385b820055e791d2567deeffca/src/camera.cpp#L320

EugeneDMyers, 2020-03-02:
Couple things I think you need to check:

(1) I did the network update.  It worked okay, but I had a copy of botui in both /usr/bin and /usr/local/bin.  I did not check to see if they were the same, but depending on how botui is started you could get the wrong one.  There appears something is causing botui to be installed in both locations.  I believe that it should be in /usr/local/bin - but you need to check the installation code to ensure that we have only one installed on a wallaby.

(2) I first used the botui repository under ~/test/botui.  However, I found that when I did a “make install” it did not use the one that I built (I put a buch of debug statements to trace out the problem) rather it used the one in the directory build by one of the update programs instead of the one that was built.  I noticed this because it did not output any debug statements and crashed when returning out of camera live and the date stamp was way off by days.

So, I manually moved the botui I built under /test/botui and was in its (child) install directory (I’m at work so I don’t have the exact path) to /usr/local/bin - everything worked properly.

(3) I then forked a copy KIPR/botui into my github account and the cloned that into my wombat, built it and made sure that the botui that it build was placed into /usr/local/bin.
     I tested and camera live worked okey, took out the model=BLACK_2017 and that worked as well.

So, we need to make sure we are using to correct botui.  I spent two nights on this until I realized that the correct one was ->NOT<- being installed. :-( 

     gene

EugeneDMyers, 2020-03-02:
A little mystified at why the inclusion of module=BLACK_2017 makes work. The open call should have defaulted to the Device class open defaults, which would make that state redundanc. Suggest that you recompile botui using the master branch libwallaby headers.  One of the problems last year was a header miss match between the libwallaby headers and botui.

prithvirajkadiyala, 2020-03-03:
I did recompile both Libwallaby and BOTUI from the master branches from github and deleted the botui from **/usr/bin** and **/usr/local/bin** before compiling and installing.

That did solve the issue and the fix you provided is working perfectly fine. I will have Tim and Steve have a look at it and test a little more on it tomorrow.

We will have to look into the install path for the **/usr/bin/botui** and make sure that we do not have multiple installs simultaneously.

Thanks for pointing out the issue with multiple installs, I didnt think it would make an issue with the custom make and install I was doing.

commit ab47eec7 (2020-03-02): Merge pull request #108 from EugeneDMyers/master

Fix double release (Resolves #107 Camera live crashes botui)
closed 2020-03-03 by prithvirajkadiyala