kipr/libwallaby #109: Move from Python2 to Python3

prithvirajkadiyala, 2020-03-03:


Zacharyprime, 2021-07-08:
I have made changes to harrogate and libwallaby that will make this work.

I don't think it's clean enough to push to master or anything, but it's there for people that want it.

[Harrogate Branch](https://github.com/kipr/harrogate/tree/python3)
[Libwallaby Branch](https://github.com/kipr/libwallaby/tree/python3)

When we move to python 3 with master, we should probably add a dropdown setting or something that allows the user to switch between 3 and 2 or make it part of the dropdown for language select.

Zacharyprime, 2022-11-04:
I'm a little more experienced with modern python than when I wrote that last comment.

I suggest sticking to python3, the community is slowly moving away from python2. It's still relevant as many codebases rely on python2 but the future should be focused on supporting python3.

edit:
It seems we have already made the transition during the refactor, closing this issue now.
https://github.com/kipr/libwallaby/blob/refactor/binding/python/CMakeLists.txt

closed 2022-11-04 by Zacharyprime