kipr/libwallaby #110: Set button functions do not work

tcorbly, 2020-03-10:
set_a_button_text
set_b_button_text
set_c_button_text

These functions do not change the text on the buttons when running programs.

Zacharyprime, 2021-05-19:
These functions are marked as deprecated in the code...
[See IdButton::setText](https://github.com/kipr/libwallaby/blob/master/src/button.cpp)

I might try and see if I can add it back, but I'm sure there was a hurdle in that that made them deprecate it.

Zacharyprime, 2021-05-19:
From reading through the commits that led up to the state of that function, I think that this never actually worked.
It seems like there were issues getting botui to interact with libwallaby.

I suggest that we just remove them entirely

Zacharyprime, 2022-11-07:
This isn't a feature that is necessary and it is probably going to be a bit of a pain to get this to work.
We have no real use for setting those buttons, especially since it's not that hard to just put "A = <option1>" as a print statement.

I'm gonna close this issue for now, if we want to add it later we can cross that bridge when we come to it.

closed 2022-11-07 by Zacharyprime