kipr/libwallaby #147: Get the Create3 ROS2 library working

Zacharyprime, 2021-10-04:
This will most likely require getting ROS2 working on the controller, which may prove to be a daunting task.

https://github.com/iRobotEducation/irobot_create_msgs

https://github.com/iRobotEducation/create3_examples

Zacharyprime, 2021-10-05:
We may be able to reinstall ROS1 and then use a ROS1 bridge to use the Create topics.
https://github.com/ros2/ros1_bridge/blob/master/README.md

bmcdorman, 2021-10-05:
We should use ROS 2

Zacharyprime, 2022-11-04:
Instead of having 3 seperate threads all ROS2 discussion should be done here.
This issue had some info on it in case we need it for later: https://github.com/kipr/KIPR-Development-Toolkit/issues/6

Zacharyprime, 2022-11-21:
As of writing this, the Wombat-OS is still being worked on.
Once that is finished we can experiment with this, but we will need to see what the final state of the OS rework will look like, how much space it takes up, and how much space we will be working with to make ROS2 work.