kipr/libwallaby #172: STM32 flashing not working on Debian Buster

Zacharyprime, 2022-09-09:
There must be a driver that we are missing as part of the setup. 
I bet there was something modified to make it work back in development rather than something specific to jessie.

The error "unexpected reply" is not always the error, it also throws ACK errors, NACK errors, and "failed to init device". It seems to be random so it's possible there is something in the background of <=v26.1 that is activating the STM32.

 ```
pi@raspberrypi:~ $ sudo ./wallaby_flash
echo 1 > /sys/class/gpio/gpio17/value
resetting co processor...
stm32flash -v -S 0x08000000 -w /home/pi/wombat.bin /dev/ttyAMA0
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Got byte 0x00 instead of ACK
Unexpected reply from device on command 0x01

echo 0 > /sys/class/gpio/gpio17/value
resetting co processor...

```

I've tried using the installation that I saved in kipr-update and the one that is pulled from aptitude package manager.
They are both v0.5 for me, and that would be the same as found on <=v26.1

I've tested it with two different Wombats so it is unlikely to be faulty hardware.

EugeneDMyers, 2022-09-15:
Not sure what is happening here as disable-bt is a remapping (include) of pi3-disable-bt.  Will look into the setup trace when I can get a chance to see why the mapping comes out different.

One thing you could try: in wallaby_flash change 'DEV=/dev/ttyAMA0' to 'DEV=/dev/ttyS0'  In theory, that should work.

Zacharyprime, 2022-09-16:
I saw comments saying to replace it with ttyS0 but I thought that it would require changing pins so I disregarded it.

I tried it but it just gave a failed to open error.
However, while doing that I realized what the real problems were.

1) The GPIO direction wasn't being set to out before editing values. 
2) The raspi-config had confusing wording. If you enable the login shell over serial, it disables the serial pins. But that option is under "interfacing" and is just called "serial port" which made me think I was enabling serial communication in general.

Zacharyprime, 2022-10-03:
I don't think I meant to close this because it's still an issue.
I'm reopening it because I think it is the cause of https://github.com/kipr/libwallaby/issues/174