kipr/libwallaby #174: Segmentation Fault when Calling Motor Functions

Zacharyprime, 2022-09-16:
This is on the refactor branch.

I get a segfault when trying to run a program containing a motor function.
It occurs when calling the function WombatDevice::transfer

```
(gdb) backtrace
#0  0x76a2852c in WombatDevice::transfer(unsigned char const*, unsigned char*, unsigned int) () from /lib/libkipr.so
#1  0x76a28254 in WombatDevice::w8(unsigned char, unsigned char) ()
   from /lib/libkipr.so
#2  0x76a26b4c in kipr::core::Platform::writeRegister8b(unsigned char, unsigned char) () from /lib/libkipr.so
#3  0x76a4282c in kipr::motor::set_motor_mode(unsigned int, unsigned char) ()
   from /lib/libkipr.so
#4  0x76a3eb20 in move_at_velocity () from /lib/libkipr.so
#5  0x76a3eb60 in mav () from /lib/libkipr.so
#6  0x76a3efbc in motor () from /lib/libkipr.so
#7  0x000105c8 in main ()
```

This is the file those last functions are defined in.
https://github.com/kipr/libwallaby/blob/refactor/module/core/src/device/wombat/wombat_device.cpp

I thought maybe it could be something wrong with the STM32 since I had some troubles with that earlier.
But analog functions are working fine so the communication with STM32 is not broken.

Zacharyprime, 2022-09-16:
Servos also cause segfault

```
0x76a2852c in WombatDevice::transfer(unsigned char const*, unsigned char*, unsigned int) () from /lib/libkipr.so
(gdb) backtrace
#0  0x76a2852c in WombatDevice::transfer(unsigned char const*, unsigned char*, unsigned int) () from /lib/libkipr.so
#1  0x76a28308 in WombatDevice::w16(unsigned char, unsigned short) ()
   from /lib/libkipr.so
#2  0x76a26d50 in kipr::core::Platform::writeRegister16b(unsigned char, unsigned short) () from /lib/libkipr.so
#3  0x76a44df4 in kipr::servo::set_servo_position(int, unsigned short) ()
   from /lib/libkipr.so
#4  0x76a44b44 in set_servo_position () from /lib/libkipr.so
#5  0x00010668 in main ()
```

Zacharyprime, 2022-10-03:
It's possible this is being caused by https://github.com/kipr/libwallaby/issues/172

Zacharyprime, 2022-10-03:
The equivalent in pre-refactor: https://github.com/kipr/libwallaby/blob/master/src/wallaby_p.cpp

Storing this for reference:
<details>
```
private:
  bool transfer(
    const std::uint8_t *const write_buffer,
    std::uint8_t *const read_buffer,
    const std::size_t size
  )
  {
    std::lock_guard<std::mutex> lock(mut_);

    struct spi_ioc_transfer xfer;
    memset(&xfer, 0, sizeof xfer);

    xfer.tx_buf = (unsigned long)write_buffer_;
    xfer.rx_buf = (unsigned long)read_buffer;
    xfer.len = size;
    xfer.speed_hz = 16000000;

    const int status = ioctl(spi_fd_, SPI_IOC_MESSAGE(1), &xfer);

    usleep(50); // FIXME: this  makes sure we don't outrun the co-processor until interrupts are in place for DMA

    if (status < 0)
    {
      logger.error() << "SPI_IOC_MESSAGE: " << strerror(errno);
      return false;
    }

    if (read_buffer[0] != 'J')
    {
      logger.error() << "DMA de-synchronized";
      return false;
    }

    return true;
  }

  int spi_fd_;
  std::mutex mut_;
};

struct WombatDeviceDescriptor
{
  typedef WombatDevice DeviceType;

  static bool isPresent()
  {
    return access(SPI_FILE_SYSTEM, F_OK) == 0;
  }
};
```
</details>

Zacharyprime, 2022-10-03:
I don't really see any obvious issues.

Possibly caused by the memset usage?
https://stackoverflow.com/questions/16793612/calling-memset-causes-segmentation-fault

Zacharyprime, 2022-10-03:
[image]

delay_usecs isn't relevant since we aren't modifying chip select.

I think the bits_per_word being set to 0 might be problematic, once this finishes compilation I will try changing that.
From what I gather the Pi only supports 8 bits anyway.

bmcdorman, 2022-10-04:
What line is the segfault happening on? You'll need to compile libwallaby with `-g`

bmcdorman, 2022-10-04:
See discord for a potential fix. Haven't tried it, but that seems likely.

twells46, 2026-08-24:
Fix PR seems to have been merged.

closed 2026-08-24 by twells46